leodesigner/powmr_comm #3: Split RS232 and ESP code

davidebeatrici, 2023-05-29:
I have a server that is running right next to the inverter, it would be really nice to be able to connect the serial port directly to it using something like https://gitlab.com/Teuniz/RS-232.

Splitting the code into separate classes would definitely help to implement that, among other advantages.

leodesigner, 2023-05-29:
Thanks for suggestion. 

I wanted to avoid coupling any other external device (orangepi, raspberrypi).
The inverter is quite noisy in RF spectrum and may cause these devices to work less stable. 
Especially with the common ground.

I guess we have a lot of other options to communicate over modbus from linux machine.

davidebeatrici, 2023-05-29:
Thank you for the fast reply.

My idea was to put at least one filtering capacitor between inverter and computer, but the devices cannot be fully decoupled. An ESP8266 can be sacrificed in case something goes wrong as it's very cheap, a high-end server on the other hand...

I think I'll design a PCB on EasyEDA.

davidebeatrici, 2023-06-05:
I completed both schematic and PCB, here's a preview of the latter:

<details>
<summary>Pictures</summary>

[image]
[image]
</details>

<details>
<summary>Components</summary>

<table>
<tr>
 <td> Name
 <td> Part
 <td> LCSC part
<tr>
 <td> C1, C2
 <td> GRM21BR61E226ME44L
 <td> C86816
<tr>
 <td> C3, C4
 <td> GRM21BR60J476ME15L
 <td> C77072
<tr>
 <td> C5
 <td> GRM155R71H332KA01D
 <td> C85963
<tr>
 <td> C6, C7, C8, C9
 <td> GRM155R71C104KA88D
 <td> C71629
<tr>
 <td> ESP32
 <td> ESP32-C3-WROOM-02U-N4
 <td> C2926676
<tr>
 <td> ESP32_BOOT
 <td> TSG002A05018A
 <td> C2888616
<tr>
 <td> ESP32_UART
 <td> PZ254V-11-04P
 <td> C492403
<tr>
 <td> L1, L2
 <td> DFE252012F-1R0M=P2
 <td> C435392
<tr>
 <td> PSU
 <td> TPS62180YZFT
 <td> C73980
<tr>
 <td> R1, R3
 <td> CRCW0402470KFKED
 <td> C482181
<tr>
 <td> R2
 <td> 0402WGF1503TCE
 <td> C25755
<tr>
 <td> RS232
 <td> R-RJ45R08P-A009
 <td> C2902700
<tr>
 <td> RS232_UART
 <td> MAX3232IPWR
 <td> C113205
</table>
</details>

The power supply is definitely overkill for our needs, but more stuff can be connected to the board by simply soldering wires to it.

I will make the order on JLCPCB, test the hardware and then publish the resources under an open-source license.

davidebeatrici, 2023-06-16:
The PCB(s) have arrived!

<details>
<summary>Pictures</summary>

[image]
[image]
</details>

davidebeatrici, 2023-06-29:
Alright, the hardware is working as it should! It took me a while to test as I had to fix a few things. I will publish schematics and PCB soon.

> The inverter is quite noisy in RF spectrum and may cause these devices to work less stable.
Especially with the common ground.

You were absolutely right: when powered by the RS232 port, simply connecting the ground from the ESP32's UART header to a CH340 interface causes the computer's USB port to reset continuously... any ideas on how to fix that?