leodesigner/powmr_comm #7: Problem with parallel model

damihighlander, 2023-09-28:
Greetings, a big thumbs up for this project.  I have an issue with inverter model POW-HVM5.5K-48V-LIP  (with parallel) - according to powmr website it also supports rs232 wifi dongle however for some reason i cannot get any response from modbus protocol, i checked the rs232 connections with an multimeter - it seems to have good connection - baudrate is set to 2400 n 8 1bit, Modbus ID Setting (setting 25) is default set to 001, i used ModScan32 for protocol testing (address 0001, id 1 lenght 100) - still nothing. Could you guide me in reading registers from this device? I have some experience in reading data and sending commands via mqtt to VMIII and MAX II hybrid inverters (if someone needs help with these particular models) powmr seems to be diffrent atleast in rs232 modbus.  Regards

leodesigner, 2023-09-30:
Do you have the original wifi plug?
It would be much easier to sniff communication between wifi plug and your inverter.
This way you will find out protocol and Modbus id as well.
At least I used this approach for my inverter.

You might want to explore this documentation for different protocols here:
https://github.com/jblance/mpp-solar/tree/master/docs/protocols

damihighlander, 2023-09-30:
Hello - thank you for the answer, no ufortunately i dont have original wifi adapter, however in the mean time i managed to get some data from the inverter. It looks like my model works on modbus ID: 1, 9600 Baudrate - i found a working list of registers from a diffrent model -> https://github.com/syssi/esphome-pipsolar/files/9657007/Protocolo_SMG.II.5KPRS232.V1.0.pdf, it seems most of these values are correct, i didnt check them all yet but having my experience with EASUN models  im pretty sure POWMR engineers keep the protocol standards aswell. For now im trying to remotely set working mode (SBU,SUB,SOL). I have a question - did you manage to change inverter working mode via modbus on your device?

leodesigner, 2023-10-04:
Yes, changing modes works as well. But keep in mind that you need to receive a valid confirmation via modbus (error codes) in order to make sure that command passed correctly into inverter.

The rs232 communication is quite unreliable due inverter RF noise. Especially during battery charging.

leodesigner, 2023-11-04:
SUB/SBU mode switch sometimes works only when you first charge battery to predefined value (see inverter instruction).
I also change inverter modes on specific predefined time.

damihighlander, 2023-11-07:
Im using nodered dashboard to visualize everything + the scheduler works pretty good right now - only few errors so far. Im stuck on changing  battery charging current parameter, it looks like either modbus register for this particular parameter is diffrent than in the manual i linked above or maybe my input string value is not correct, i know this may be a broad question but maybe you could help me find an error - when changing work mode i simply connect to modbus and send a simple string (0,1,2,3)  on address 301 - this works fine, however for some reason changing "Maximum charging current" wont work 

Modbus specs from pdf file: 
Maximum charging current, 
Unit: 0.1A, 
Data format: Uint,
Initial Address: 332,
Number of registers: 1,
Read: R/W 

and this is my error from debug node: msg : error
"Error: Quantity should be less or equal to register payload array length: 2 Addr: 332 Q: 1"    what should an "inject payload" should look like here?

leodesigner, 2023-11-07:
Make sure that you have a correct byte ordering for two bytes integer value (big endian, little endian)
https://en.wikipedia.org/wiki/Endianness

You need to look closer what is the Node-Red function expects.
Also some inverters accept only predefined values like 10a, 20a etc.

Try to read this modbus register first and figure out what you get in response.

akosiramnad, 2024-11-03:
hi @leodesigner,  how can i sniff communication between wifi plug and inverter?
can you give me instructions and tools needed to sniff. thank you.

leodesigner, 2024-11-04:
@akosiramnad the easy way would be to flash esphome basic firmware with serial debugging on.
If you have an original wifi plug which works with your inverter, you would need to sniff communication on both lines RX and TX. This way you can grab what your wifi plug requesting from the inverter. (state and commands).

The other approach would be to create a custom firmware for esp32/esp8266 and sniff all serial communication between Wifi plug and inverter. For most popular inverters you can find already some information online.

akosiramnad, 2024-11-05:
> @akosiramnad the easy way would be to flash esphome basic firmware with serial debugging on. If you have an original wifi plug which works with your inverter, you would need to sniff communication [...]

I have the original wifi plug, can you please guide me step-by-step or give me links to do the sniffing.Thanks

leodesigner, 2024-11-06:
@akosiramnad I am a bit busy at work now.

These links might be helpful for you:
https://community.home-assistant.io/t/sniffing-rs232-comms-between-sph6000-and-a-shine-s/553421
https://community.home-assistant.io/t/rs232-logging-of-well-water-level-monitor/464952
https://community.home-assistant.io/t/how-to-recieve-rs232-data-in-esphome/595602
https://www.reddit.com/r/PLC/comments/sthds7/best_way_to_snoop_on_rs232/

akosiramnad, 2024-11-08:
> @akosiramnad I am a bit busy at work now.
>
> These links might be helpful for you: https://community.home-assistant.io/t/sniffing-rs232-comms-between-sph6000-and-a-shine-s/553421 [...]

Thank you very much, this would be a great help for me.