leodesigner/powmr_comm #8: Esp8266 power pins

jayprojects, 2023-10-24:
Hi thank you for this great project. 
i just want to clarify the hardware part little bit. you mentioned 

```
The hardware part is simple esp8266 module with RS232 level converter attached to the
    #define SS_TX_PIN D8 // GPIO15
    #define SS_RX_PIN D7 // GPIO13
```

But what about v & ground pins? As i understand V pin (blue) on the RJ45 is 12v. have you used a step down circuit to make it compatible with esp8266? Also for the TX/RX do we need a logic converter? 
Thanks again.

leodesigner, 2023-10-24:
Hey,
Yes, I used step down module to convert from 12volts to 5v.
Please check esphome based project for more detailed information regarding hardware part:
https://github.com/odya/esphome-powmr-hybrid-inverter

jayprojects, 2023-10-31:
Apology for delayed follow up from my end. didn't had time to re-visit on this last week. 

so on esphome project, it appears 12v positive pin was not connected. and no logic converter was used. (esp8266 is powered by other source). it only used rs232 to ttl converter and directly connected PowMr to ESP32 (in our case esp8266).

I tried the same setup, and i found no activity on the serial communication. 

Maybe im missing something here, or maybe serial communication need to be "turned on" on the inverter. or Indeed a logic converter is needed on the PowMr side in addition to RS232<->TTL converter as my original question.

Bit off topic, but just fyi, My inverter is POW-HVM55.k-48v-N. I also tried using a USB-to-RJ45-Cable-for-RS232 and tried to read it using a windows laptop directly. I did not detect any data on serial port either, the adapter cable on the usb end got supper hot!

leodesigner, 2023-11-04:
The inverter will only respond to a valid modbus frame request.
You can power your esp8266 from any source you prefer (via DC-DC step down buck converter or external power source)
You will need a RS232<-> ttl converter to connect your esp8266 board to the inverter.

You can also use some kind of external software to request data from inverter as long as you can send a valid modbus request via rs232 interface.

jayprojects, 2023-11-06:
Thanks for following up. unfortunately  im still stuck. 
You are probably right maybe my modbus frame request is wrong.

So at this point im trying to get some (any) response from the inverter. your main.cpp is large. as it has many features.

I strip it down to a hello world type

would you please have a quick look and let me know if you spot that i have missed anything.

```
#include <SoftwareSerial.h>
#include <ModbusMaster.h>
ModbusMaster node;

#define SS_TX_PIN   D8
#define SS_RX_PIN   D7

SoftwareSerial Ser1(SS_RX_PIN,SS_TX_PIN);

void send_request() {
  uint8_t j, result;
  #define MAX_MBUS_WORDS  50
  uint16_t data[MAX_MBUS_WORDS];
  
  #define MAX_MBUS_PKT_S (MAX_MBUS_WORDS*2)
  
  static uint8_t charArr[2*MAX_MBUS_PKT_S + 1]; 
  uint8_t *myPtr;
  myPtr = charArr; 
  
  uint16_t nregisters = 45;
  
  result = node.readHoldingRegisters(4501, nregisters);
  
  // do something with data if read is successful
  if (result == node.ku8MBSuccess)
  {
    for (j = 0; j < 45; j++)
    {
      data[j] = node.getResponseBuffer(j);
    }
    Serial.println("Got some response ");
  }
  else{
    Serial.println("Node not success");
  }
}

void idle() {
    yield();
}

void setup() {
    Serial.begin(115200);
    Ser1.begin(2400);
    Ser1.enableIntTx(false);
    node.begin(5, Ser1);
    node.idle(idle);
}

void loop() {
    Serial.println("Loop ");
    send_request();
    delay(3000);
}

```
As mentioned the goal is to "get some response". and maybe print out data array.

i tried that with esp8266 as well as a D1 Mini. in both case, always get "Node not success"

I tried two Max3232 board from the same packet. i can't say for certain that they are good. so there is that. 

Here is the setup. As mentioned its a POW-HVM55.k-48v-N inverter.

[image]

leodesigner, 2023-11-08:
The code looks valid.
Are you sure that your inverter supports modbus? 
Maybe it will respond to pipsolar protocol?

Take a look here:
https://github.com/jblance/mpp-solar/tree/master/docs/protocols

You might want to send QPI<cr> command and see if you get any response.

jayprojects, 2023-11-13:
I contacted to powmr support. they confirmed it is modbus.
they provided this modbus configuration 
[BMS_Modbus_RTU communication description.pdf](https://github.com/leodesigner/powmr_comm/files/13340214/BMS_Modbus_RTU.communication.description.pdf)
. not helping. coz im failing to even make the handshake.
I tried few other usb devices with pc. all failed. at this point im thinking maybe the com port on the inverter is bad!

leodesigner, 2023-11-16:
It's really hard to diagnose what is the issue, 
Looks like they provided BMS modbus registers, not an inverter. And it's not clear who supposed to poll BMS data.
You might want to check this project for your inverter as well:
https://github.com/syssi/esphome-smg-ii