leodesigner/powmr_comm #9: Powmr  4.5kW support?

vpritulin, 2024-01-05:
Hello! I have 
Hello! I have a Powmr POW-HVM4.5K-24V hybrid inverter. But I don’t quite understand whether this assembly will work for him or not? 
https://powmr.com/products/hybrid-inveter-charger-4500w-220vac-24vdc

wifi module which is used https://powmr.com/products/powmr-wifi-module-wifi-4-5k-6-5k

leodesigner, 2024-02-29:
Check this repo: https://github.com/leodesigner/powmr4500_comm

manfredr74, 2024-02-29:
... have found that source and there is a possibilty to read via ModBus and convert.Might be more interesting flashing the WiFi-Dongle with Tasmota- Software and a configuration of Tasmota to receive all data via MQTT.-- Manfred ReisnerCEO, Smart Mobility, System Safetye-mail: ***@***.***: +4368181113224ACTARON GmbHHauptstrasse 297441 PilgersdorfAustria

leodesigner, 2024-03-04:
For now you can use custom include file provided with all registers.
There was an experimental project for ESPHOME, but afaik it wasn't finished.

vpritulin, 2024-05-31:
@leodesigner 
Hello! 
If I understand correctly, this repository does not support homeassistant and esp8266. And you need to use the file include/inv8851.h as a library yourself and then work with it?? Do I understand correctly?

https://github.com/leodesigner/powmr4500_comm

leodesigner, 2024-06-04:
> @leodesigner Hello! If I understand correctly, this repository does not support homeassistant and esp8266. And you need to use the file include/inv8851.h as a library yourself and then work with it?? Do I understand correctly?
> 
> https://github.com/leodesigner/powmr4500_comm

Yes, the header file is for development, there is a project for ESPHOME and HA (not sure about the readiness) in issues.

vpritulin, 2024-06-04:
@leodesigner 
Thank you! Maybe you know how to find a register map for daxtromn 10kW?

leodesigner, 2024-06-04:
@vpritulin no, but might be compatible with PowMR 10kW probably.