leodesigner/powmr_comm #11: Support for POW-HVM5.5K-48V needed

Schekamatik, 2024-03-18:
Since some days I am trying to get data by RS232-to-TTL module (MAX3232CSE). I am not getting any response from the inverter. 
I have used this example code: 
https://github.com/leodesigner/powmr_comm/blob/main/src/main.cpp

The connection to esp32 was done like it is described here:
https://github.com/syssi/esphome-smg-ii

Can someone please support me?

Thank you in advance.

leodesigner, 2024-04-01:
This is absolutely different beast, check this repo:
https://github.com/leodesigner/powmr4500_comm

PetroPasichnyk, 2024-04-25:
I have got a POW-HVM5.5K-48V-LIP and managed to receive a quite good result with https://github.com/syssi/esphome-smg-ii 
RS232-TTL MAX3232 --  ESP8266 -- Esphome -- Home assistant
Only issue was 8266 has a little memory, so, only essential sensors have been connected, approximately a half of all available.

kevinari26, 2024-12-19:
hi, I am using [POW-HVM6.2M-48V](https://powmr.com/products/hybrid-inverter-charger-6200w-220vac-48vdc)
and ESP32, I am using UART 2 (GPIO 16, 17)
the response I get always `ku8MBResponseTimedOut`
can you help me @leodesigner or anyone?