leodesigner/powmr_comm #12: Pintout and voltages from RJ45 port on POW-HVM3.2H-24V-N

Oxmaster, 2024-03-21:
I've got POW-HVM3.2H-24V-N, still waiting for the batteries, so testing it with benchtop power supply.
[image]

I'm checking the RJ45 pinout on my bench and I'm getting different outputs than from the diagram provided. 

[image]

White green (1) is GND
Brown (8) is + VBat
Blue (4) is + 13.2 V

or at least that's what my multimeter says. Don't want to remove the top cover because of the tamper seal to check for the RJ45 board.

leodesigner, 2024-04-01:
Maybe you got a new version. Please check twice before connecting anything.

closed 2024-04-03 by Oxmaster

serhii-nakon, 2026-10-03:
Hello @Oxmaster .
Does it works? I have exactly the same inverter and would like try this project to pull data from inverter