leodesigner/powmr_comm #14: Add support for HHJ60-PRO

ss89, 2024-04-30:
Mid of last year PowMr released the HHJ60-PRO.

Can it be supported as well?

Refs:
* https://powmr.com/products/mppt-solar-charge-controller-for-parallel-60amp-12v-24v-36v-48v

leodesigner, 2024-05-07:
Sorry, but I don't have it.

vpritulin, 2024-05-31:
@ss89 Hello! you can try to do something similar as this video explains. 
https://www.youtube.com/watch?v=hQCPfaHJI6o

ss89, 2024-06-01:
> @ss89 Hello! you can try to do something similar as this video explains. If your MPPT has RS232 output https://www.youtube.com/watch?v=hQCPfaHJI6o

Is the code somewhere on github? I don't feel like copying the code byte by byte

vpritulin, 2024-06-01:
@ss89 yes, you can check this repo.  If anything, 
https://github.com/W1zzard/solar-mon 

I am not the author of this video and repo.